basilfx/TRADFRI-Hacking #1: Serial number?

OleUrgast, 2017-11-26:
Hello,
its a great, that you can read and write the firmware via jtag. Many thanks for your work.

Do you know, where the serial number is coded? The hardware of warmwhite and whitespektrum Tradfri-Zigbee-module is the same. I have build lightstrip-controllers using the tradfrimodule and LogicLevel MosFETs. The most expensive part is the Tradfri-Bulp, making a whitespectrum controler twice as expensive as a warmwhite only.

So the cheapest way to build multiple whitespectrum controler would be to read the flashmemory from a whitespectrum and flash it to a warmwhite module taken from a cheap GU10 warmwhite. Of course, before flashing you have to change the serial number / mac to avoid confusion in the zigbee network.

The second use of your research is to clone tradfri remotes. The big disadvantage of Tradfri is, that you can only use one remote control for a bulp. If you connect a second one, the first one gets lost.
So if you connect a remote to the bulps of a room, cloning the complete flash (firmware and settings) to another remote will alow to use both remotes for the same lights. So I can use as many remotes in a room as I want. Here of course the serialnumber has to be the same.

Will the cheap jtag adaptor (about 3$ at eBay) do for flashing tradfree? or have i to use something special?

basilfx, 2017-11-27:
Hi @OleUrgast,

There are two possibilities: either the serial number is depends on the MCU, or it is embedded in the SPI flash. If it's the first, than every module will have its own unique serial number, which doesn't depend on the firmware. If its the second option, then you'll need to change that.

I haven't figured it out yet (working on the radio driver first), but I have used [this](https://github.com/basilfx/RIOT-applications/tree/master/spi_flash_identify) application to read the flash ID.

Regarding flashing: as long as it supports SWO/SWDIO/SWCLK + Reset, than it should work. I have used a STK3600 development board [set to debuggin mode](https://www.silabs.com/community/software/simplicity-studio/knowledge-base.entry.html/2014/11/24/using_an_efm32_start-H9Gy).

badblocks, 2017-11-30:
@OleUrgast:
You can have 2 remotes controlling one/group of lamps; learn one remote to a lamp/group of lamps and reset the second remote and then hold both remotes together and pressing both buttons inside the remote. Then they both controlling the same lamp/group of lamps.

OleUrgast, 2017-12-22:
@badblocks:
I tried that, but it didn't work.

The reason seems to be: To do the trick you described, the first remote not only have to be connected to some bulps, but also to a Tradfri gateway.
I use a hue bridge. The remote got it's zigbee settings via touchlink, so it can be used in parallel to the bridge.
In this configuration, the trick to copy the settings from one remote to another seem not to work. Its not surprising, as I discovered it seems impossible to touchlink a hue-connected Tradfri remote to an not already hue-connected bulp also. 

So firmware/settings cloning via SWD seems the only option to use two or more Tradfri remotes in an hue environment.

spacesloth0x, 2017-12-27:
@basilfx Thank you for the effort you're putting into this. I read in the RIOT-OS pull request that you managed to read the flash. Would you be able to share that dump? It would seem the bulb state is stored in there, perhaps more.

basilfx, 2017-12-27:
@spacesloth0x I did not have read the flash contents yet. I only read the flash ID, to identify the chip used. But I presume it is just a standard flash chip, so you should be able to read it straight away.

darconeous, 2018-06-12:
@OleUrgast wrote:

> The big disadvantage of Tradfri is, that you can only use one remote control for a bulp. If you connect a second one, the first one gets lost.

I believe you are mistaken. I am currently controlling my ZigBee Light Link lights with multiple unmodified TRÅDFRI remotes. You can set this up using the Ikea TRÅDFRI bridge and the app (by moving the remote into the group), or you can do it when you commission a new remote:

1. Make sure you are starting with a TRÅDFRI remote that is already paired with all of the lights that you want to control.
2. Take the NEW remote that you want to add to the group and press the button four times. The LED will pulse six times.
3. Wait 10 seconds.
4. Press and hold the pair button on both remotes at the same time. The red LED on *both* remotes should start to glow or flash. Keep holding down the buttons until both lights finally go out—it will be around 10-15 seconds.

That's it. The remotes are now a part of the same group. Either remote will now control all of the lights. Removing or adding a light to either remote will automatically add the light to the other remote.

No need for all of the serial number hacking nonsense, which would almost certainly not do what you want.

OleUrgast, 2018-06-12:
As I said: I use a hue bridge. So I have to connect the remotes to the hue hub first, then touchlink  remote and bulb. Else touchlink a bulb to the remote will result in the bulb loosing connection to the hue-bridge.
But after connecting the remote to a hue bridge, touchlink seems to work with devices already connected to the hue-network only. Trying to touchlink any new or reseted device to the hue-bridge connected remote do not work. You have first to connect the targed device to the hue-net also. Than - and only then - you can connect hue-paired Tradfri remote and the hue-connected bulb by touchlink.
The same with a second remote: If you press the button 4 times, you reset the second remote - so its not connected to the hue-bridge. It will not react to touchlink from a hue-connected tradfri remote.

I tried the described way of copying a tradfri remote many times. Using a tradfri gateway it works fine. If a hue-bridge is used, it does not work at all.

Using the tradfri gateway is not an option to my setting, as in some cases I need the hue-dimmer for scenes (like the simple scene "all light off" set to the dim-low button at the dimmer beside the entry of the house, but also to select diferent colour-settings on the hue-lightstrips by pressing "on" multiple times) and to use the much better hue motion sensor.

As I already set up my complete house with a mix of tradfri remotes and hue-dimmers (only one tradfri remote in a group), I have not tried copying the flash yet. But next year I will upgrade my guesthouse also to zigbee - then I will giving copying the flash a try, as the tradfri remotes are less expensive and allow to control the colour (on Ikea colour bulbs) and the lighttemperature much more confortable as the hue-dimmer.

darconeous, 2018-06-12:
Trust me, I know what I’m talking about. I use a Hue bridge, too. Pressing it four times will cause the remote to forget the network. Pairing it back to another remote that is on the network will bring it back on the network AND put the remote on the same group. I’ve done this dozens of times. 

I swear it works. I use several TRADFRI remotes with my Philips Hue setup, including in the configuration you say doesn’t work. My working setup is proof that it does work. I can make you a video if you still do t believe me. 

Perhaps your remotes have an old firmware?

OleUrgast, 2018-06-12:
I see the missunderstanding.

You pair a remote (in german named "Fernbedienung" and a dimmer (in german simply called "Dimmer"). That works without issues in my setup also. It even works to pair dimmer and hue-bridge, remote and hue-bridge, than touchlink dimmer to light 1 and 2 and remote to light 2 and 3 - so light 2 is controlled by both controls (witch didn´t make sense with a remote, as it tongles so using the dimmer to turn on, it will result in light 2 going off and light 3 on when you tongle with the remote afterwards).

The problem is to pair two remotes ("Fernbedienungen" - the ones with the buttons) in a hue setup.

As I last used the Tradfri Gateway (it was about december I think), the number of the dimmer firmware was just a little bit higher than the one available for the remote. I never tested if there is a new firmware for the remotes or the bulbs since December, as my tradfri gateway got damaged (simply to repair, the reset-button crashed and triger a permant reset - but I have to find a fitting replacement and have no need to do so, as I found no news about newer firmwares for remotes or bulbs since last year - only the tradfri gateway itself was updated).

Was where any firmware updates to the remotes since december? Have you ever tested pairing two remotes, not only remote and dimmer?

I do not like the dimmer - als I nearly everythere use lights with adjustable colourtemperature. In my opinion it is also less comfortable to simply switch the light on and off (which is the most used scenario)

But you give me an idea: Maybee I try to pair the lights to a dimmer first, then cloning the settings to the first remote, than to a second one. Then resetting the dimmer. But I am not totally hopefull, as I already tried two dimmer and two remotes (connected to hue bridge) with a buld (already paired with hue-bridge also). Touchlinking dimmer-dimmer-remote to the bulb works. Touchlinking dimmer-remote-remote results in only the second remote (and the hue-bridge) controling the light; dimmer and first remote did not control the bulb any longer.

So dimmer and remote behave totally different in touchlinking...

darconeous, 2018-06-12:
> Have you ever tested pairing two remotes, not only remote and dimmer?

Yep, and it works the same way for me.

> EDIT: YOU MADE MY DAY !

Yay!

> As I said, copying setting via touchlink from a remote to another remote did not work. Also copying settings from a remote to a dimmer did not. Also touchlinking a reseted bulb not in the hue-net by using a hue-paired remote didn´t.

I have a lot of trouble getting touchlinking to work, but it looks like you are having a harder time than even I am.

One trick that seems to improve the chances of it working is to press&hold the button of the assisting remote (the one already on the network) a fraction of a second before pressing&holding the button on the joining remote (the one you are adding to the group). Not sure if it is a placebo, but it seems to improve my chances.

Another hunch I have is that the Ikea remotes/dimmers are easily confused by other on-network traffic. I seem to have much better luck adding remotes and lights when I am away from my home network, or (presumably) if I have turned off all the other devices on the network.

Worth a try. Just take two remotes to work or around the block and try pairing them out of range of your network.

OleUrgast, 2018-06-12:
Yes, pressing the assisting dimmer button first (for a fraction of a second) works best. It is exactly the opposite of touchlinking a dimmer to the hue-bridge: There I have to press the dimmer button a little bit earlier as the "sending" hue bridge (I use the android app hue essentials for starting touchlink).

I now tried to connect an "unused" bulb (self-made tradfri lightstrip controller whitespectrum) using the dimmer. It worked. Simply connected the "bulb" to the dimmer using touchlink, then searched for new lamps in hue essentials - and it was found (as a GU10 colour spectrum bulb as this was the source I got the zigbee-controler from). Verry easy - I do not have to transport all new lamps to the hue-bridge any longer.
An other reseted bulb (prior already connected to hue) got touchlinked to the dimmer (and the remotes in the same group) but was not discovered by the hue bridge first. It seems it serial was "remembered" so it was not new, even as I had discarded it before. But after restarting the bridge it was discovered as new, so no real problem.

I may have found the reason, why copying remotes works for you but not for me. I remember last year the Tradfri-Gateway tried to update remotes with the same(!) firmware-version it already had. Maybe a "silent" update - some othe companys did simular in the past, updating the "same" firmwarenumber fixing minor bugs. But on my remotes this mostly failed. I did not thought about it much, as after the next firmware update the gateway got, it showed all remotes as having actual firmware...
I got no new remotes after that, so maybe while showing the actual firmware number they realy have a buggy pre-version.

So the "second" use for firmwarecloning (my first post) is gone. The first one stays: cloning the firmware from a whitespectrum zigbee-module to a cheap single-colour module - to make ww/cw lightstrip controller cheaper. But I already have build three controler, using only one now, so this it not urgend. It will get interesting, if my selfmade cheap Big-Fload-Project will work (using 4 normal cw-60x60 panels, 2 warmwhite, two coldwhite, building one big 130*130 panel (10cm space between the 60X60 for cabeling and electronic), as I will need at least 5 of the "Big-Float". But at the moment most of my time is spend to other projects

darconeous, 2018-06-13:
I'm happy I was able to help with at least one of your roadblocks!

> The first one stays: cloning the firmware from a whitespectrum zigbee-module to a cheap single-colour module - to make ww/cw lightstrip controller cheaper.

It's not clear to me what you are asking for, since it would seem that there is no longer any need for the location of the serial number. Are you asking for a copy of the color-temperature firmware, or are you asking for instructions on how to extract the firmware from (and subsequently reprogram) the modules? Note that it would likely not be legal to post an extracted firmware image on the internet, due to copyright.

I remember hearing from someone that IKEA did not apparently set any of the protection fuses, allowing the firmware to be read straight out of the chip—which is very surprising. I don't know if that is still the case. If I was IKEA I would lock that down ASAP.

----------

As a side note... [IANAL](https://en.wikipedia.org/wiki/IANAL), but I am suspicious of the legality of (from a copyright infringement perspective) of extracting proprietary firmware from one piece of hardware you own and subsequently running it on different hardware not intended to run it—even if the different hardware was purchased from the same vendor. Then again, I doubt IKEA would care unless you somehow started eating significantly into their profit margin (doubtful).