basilfx/TRADFRI-Hacking #2: "Devboard"'s Components' Values/Specs

git-marc, 2018-01-31:
Ref.: "Devboard"'s Components' Values/Specs

Dear Bas,

Thanks for all the information and files on TRADFRI-Hacking (Github):
An extremely helpful Package!

I would like to play & explore TRADFRI using your "devboard" schematics
but i did not succeed in finding further information on the components' specifications.
(“Devboard.kicad_pcb”, “Devboard.sch” and  their "Devboard bom file.csv” export did not reveal a hint)

May i ask which values you would recommend for the capacitors, resistors, switch?

I hope that you may find a spare second to reply
& look forward to hearing from you!

Best Regards!

git-marc, 2018-01-31:
...
"devboards" schematics connections using "Fritzing" ...
[https://picload.org/view/ddppoool/inquiry_tradfri-devboard.jpg.html](url)
...

basilfx, 2018-01-31:
The values are not very specific: take around 10µF for C1, 100nF for C2 and 1K for R1. They should work.

git-marc, 2018-02-01:
Thanks a lot for your reply, will try it!
...
updated “devboards" schematics (via “Fritzing”):
picload.org/view/ddwggoci/inquiry_tradfri-devboardok.jpg.html
_(link only seems to work when copy/pasted into browser's address field)_
...

closed 2018-02-01 by git-marc