basilfx/TRADFRI-Hacking #5: Output via UART

jonathanschneider, 2018-03-03:
Hi,

I have a dimmer with an ESP8266 which I would like to replace with the ZigBee module I took out of a Tradfri bulb. Does the Tradfri module output anything via UART? If so, which pins are used for UART?

Any help would be appreciated.

sorenfalch, 2018-03-04:
Hi Jonathan!

I'm thinking about doing the same thing (have also bought a couple of GU10's for the same purpose).
I found this video which I found quite informative. Maybe it can also help you:

https://youtu.be/bNS_fINyaQE?t=43m15s

BR, Søren

basilfx, 2018-03-05:
As far as I know, the pins do not output to UART. What makes it harder is that the EFR32 have flexible pin mapping: any GPIO could be UART :-)

closed 2020-01-11 by basilfx

MattWestb, 2021-06-22:
@jonathanschneider If you can flashing one new bootloader then you can flashing one Zigbee Controll Bridge firmware and you have CLI for sending and receiving all commands you like then the device is in one network.

Example: Putting one "on" command in the buffer (ZCL command). Then sending the buffer to group 0x0006 (mulicast send command).