basilfx/TRADFRI-Hacking #6: Making a commit (LED module teardown)

ludwo88, 2018-05-29:
Hi @basilfx,
I would like to make a commit to your project. I made a teardown of the 30W Trådfri LED driver module. So far I have a bunch of pictures and made some basic research on the components used.

I am not able to create a pull request or upload files. Could you make a branch for me? Thanks

basilfx, 2018-05-29:
Hi @ludwo88, you should be able to fork this repository, clone that fork, make the changes and push it. Once you visit this repository, it should propose you to create a PR. There should be some help on [this](https://help.github.com/articles/creating-a-pull-request/) page.

ludwo88, 2018-05-29:
Great, after the fork I was able to make changes and GitHub offered to create a PR which now shows up here as well. Thanks for the tip. I am quite new to git repositories...

closed 2018-05-29 by ludwo88