basilfx/TRADFRI-Hacking #11: Antenna of breakoutboard ON breakoutboard

oliv3r, 2018-06-04:
Looking at the latest 3D rendering on the breakout board for the ZLL module, I notice the antenna is sitting ontop of the PCB. Would it not be more helpfull to move the antenna offboard, as is done with the led drivers? (e.g. have it stick out/no PCB underneath). The only real blockage would be the reset button which in turn could move down a little bit between the pin-headers or to the bottom left where the led also lives.

Further more, would it be a good idea to put through-hole pads underneath the ZLL module? E.g. that way we can either solder the module directly onto the carrier board, or solder pinheaders onto the ZLL module and headers onto the carrier board to make it swapable (or even put pins on the carrier board and 'squeeze' the ZLL module temporarily in between.

oliv3r, 2018-06-11:
why not have the antenna stick out however? All the modules from IKEA also do this from images I have seen. Granted, performance is not important at all herein.

darconeous, 2018-06-11:
Copper fill below the antenna would have killed the performance. It was a good idea to remove it.

It's not clear how much having the antenna against the bare PCB would have. It would likely be minimal, but not zero. For a break-out board intended for testing purposes, I think it is more than sufficient.

Then again, I don't think these antennas are tuned with any great precision. I've noticed a very large variance of sensitivity across the handful of devices I have purchased. 

I'm curious though why you are going for this sort of design with the break-out board. Wouldn't it be nicer to have a board which you could just plug into a breadboard? You could still keep the debug headers...

darconeous, 2018-06-11:
Have a look at the copper keep-out zone from the remote:

[image]

It's much larger than your keep-out zone.

oliv3r, 2018-06-13:
I was thinking something like https://gitlab.com/olliver/TRADFRI-Hacking/raw/ICPSLC24-10NA/teardowns/ICPSLC24-10NA/controller_top_01.jpg was my thought :)

as for the pinheaders I think having it breadboardable/pluginable just depends on where you solder the pins, no?

darconeous, 2018-06-13:
> as for the pinheaders I think having it breadboardable/pluginable just depends on where you solder the pins, no?

The more narrow the board, the better it will work with a breadboard.

oliv3r, 2018-06-14:
actually, we can also find this in the FCC submitted manual: 
[image]