basilfx/TRADFRI-Hacking #24: upload OTA rgb bulb firmeware to radio modul from a remote control

NetHans, 2018-10-12:
Hello,

I have taken the radio module from a Tradfri remote control. Is it possible to play the functions of the RGB bulb on this wireless module by means of the OTA firmware? Or someone has a full dump of the firmware of the RGB bulb?

oliv3r, 2018-10-18:
I don't know what you consider 'full' dump of the firmware, but in the firmware dir there are some dumps/update files.

oliv3r, 2018-10-24:
ah right, well a few levels up is a JTAG dump of the full firmware, that bas was able to read and restore. so there's that.

oliv3r, 2018-10-24:
obtaining all firmwares from the devices is a different challange of course :)

basilfx, 2018-10-24:
I was able to dump a firmware, overwrite it with something else and load it again using the steps mentioned [here](https://github.com/basilfx/TRADFRI-Hacking#software-used). Like @oliv3r, you need to have a dump of the full firmware first ;-)

NetHans, 2018-10-24:
I have now put up for testing the firmware led1650r5-1.2.214.bin. The connected LED behaves as expected, but I can not find the chip through the gateway.

In the folder of the firmware I also found the file led1650r5-1.2.214.strings. Can it be that important configurations are included here? How can I first save the configurations of my chip and how can I then upload the led1650r5-1.2.214.strings to the chip?

basilfx, 2018-10-25:
The strings file is just a dump of all the strings that can be found (using `strings <filename.bin>`). It's of no use, apart from hacking the firmware.

You probably need the contents of the SPI chip as well. However, I never attempted to dump it.

efuturetoday, 2018-12-04:
I am also interested in a full dump of the flash and SPI-chip-dump. Same idea here: Turning the remote into a RGBW ZigBee Controller. I found an image of the PCB of the RGBW bulp: https://i.ibb.co/V3Gb5qT/3xtUrUX.png seems to use another kind of Zigbee-Module-Board. Maybe same CPU.

oliv3r, 2018-12-04:
looks like the same board but with the metal cap removed ... check out https://diystuff.nl/tradfri/tradfri-zigbee-light-link-module/ for detailed pictures ...

efuturetoday, 2018-12-04:
Agree should be same hardware - just different Layout even the GPIO pins count is the same. Just an Idea: Riot-OS is supported from now. Can we just dump the flash with a simple firmware over serial? And Flash the same dump with it Ingo a different device. After this just flash the original firmware with JTAG. Should work?

efuturetoday, 2019-03-01:
**Update:** I finally got the dimmable white bulp 1000lumen software flashed on the remote. For pairing i used this instruction: 

>  I then reset the bulb by turning it on and off 6 times. Once that was done I initiated a search for a new light from the Hue app without success. I then initiated another search via the Hue app, and while that search was running, I turned the light off and then back on again, and the Hue app found it.

I also did some flash dumps and it seems that the app just uses the simulated eeprom for persisting data. Hopefully i made someone happy out there - you can turn a 5€ remote into a hue compatible controller.

slm85kbh, 2019-10-15:
> **Update:** I finally got the dimmable white bulp 1000lumen software flashed on the remote. For pairing i used this instruction:
>
>
> I also did some flash dumps and it seems that the app just uses the [...]

I'm trying the opposite, can you elaborate on how you succeeded in this? I've tried via STLink-v2 and openocd, but it gives me an "Unknown MCU Family"

CableCatDK, 2019-10-15:
You might find the infomation in the guide usefull, even if it in Danish:
https://drive.google.com/drive/folders/1SMhPC1echgKWlc-W4N89JKZ2KiJ_H_ZD

It tells how to use JTAG to dump or flash the ZigBee module.

slm85kbh, 2019-10-17:
Tak CaleCatDK, det var en kæmpe hjælp! Det er på dit pcb design jeg roder :)
Danish was not an issue ;)
And I managed to apply the changes to a normal pc with stlink-v2, and now I have a working remote with the zigbee module from a bulb....

CableCatDK, 2019-11-30:
> And I managed to apply the changes to a normal pc with stlink-v2, and now I have a working remote with the zigbee module from a bulb....
Can you please describe how you got ST-Link v2 working. I just got one now, and I want to make a guide for windows users.

MattWestb, 2020-06-29:
Normally for bulbs share the same firmware but different model / setting its stored in the userdata that not being erased with normal internal flash erase.
From GDB:
(gdb) info mem  
Using memory regions provided by the target.  
Num Enb Low Addr   High Addr  Attrs  
0   y   0x00000000 0x00040000 flash blocksize 0x800 nocache  
1   y   0x0fe00000 0x0fe00800 flash blocksize 0x800 nocache  
2   y   0x0fe10000 0x0fe12800 flash blocksize 0x800 nocache  
3   y   0x20000000 0x20008000 rw nocache  

Reg 0 = Flash (256K)
Reg 1 = Userdata (2K)
Reg 2 = Lockbits (10K)
Reg 3 = Ram (32K)

More info with SWD flashing: [Flashing the ICC-1 Module](https://github.com/MattWestb/IKEA-TRADFRI-ICC-A-1-Module/tree/master/Flashing-MG#mem-regions)

sauer-m, 2021-08-17:
Hi there,
did someone has a complete dump of the fyrtur blind? 😬

MattWestb, 2021-08-18:
I think byes of them is not so "hacky" and they is little more expensive then the cheapest bulb.

You can trying "extracting" the firmware from the OTA file the is rapped with one signing and only need finding the start of the code part and cutting the heeding and ending signing part away then all the code is not encrypted.
I think (without knowing) that the user data is only changing the name of the device but i cant being sure (with the On/Off / Open/Close button its changing the device type and how the device is working with the same firmware).

I have seen scripts that is extracting the metadata from one OTA file and then can extracting the APP (the main flash part) from it but i cant finding it from the moment.

honglihu1986, 2022-07-27:
I have a JAZZDANS,but i don't know how to dump of the firmware, it's can't connect to BLINK with SWD.

MattWestb, 2022-07-27:
If its the classic IKEA ICC-A-1 Zigbee module is shall working OK with one no original J-Link  probe like or Black Magic Probe or other SWD probes.
Then JAZZDANS is not sold in EU i cant saying if IKEA have updating the Zigbee module to the new one (Silabs original MGM210LA22JNF2) and if its you need one real J-Link for reading and writing the flash in the chip then its have higher security that is not working with other SWD probes.

Can you posting one photo of the PCB with the Zigbe module ?

honglihu1986, 2022-07-29:
https://drive.google.com/file/d/1KmIRXP6XX3yp3b8BulPCjg8f_wANTR3K/view?usp=drivesdk

MattWestb, 2022-07-29:
Its one ICC-A-1 module (normal "old" one) so shall being easy dumping and flashing !!

06 | PF0 | SWCLK
07 | PF1 | SWDIO
11 | VDD |
12 | GND |

shall being enough.

Pin out from FCC https://github.com/MattWestb/IKEA-TRADFRI-ICC-A-1-Module/tree/master/teardowns/ICC-A-1

Always dumping the man flash (0) **and** user data (1) !!