basilfx/TRADFRI-Hacking #27: Additional item E1745 TRADFRI Motion Sensor

OneOfTheInfiniteMonkeys, 2019-10-29:
Additional item E1745 TRADFRI Motion Sensor
Uses same Zigbe board external connection points as other modules, Though later version having same CPU. Zigbe board marked Draco1.0_105°C
Uses:
  2 CR2032 batteries for 6 volt operation
  E931.96B Infra red motion sensor interface device
  LHI 968 – Sensor Motion, Pyroelectric (presumed as no markings)
  TCA9535 Low Voltage 16-Bit I2C and SMBus Low-Power I/O Expander (for buttons and LEDs)
  3 tactile switches
  5 LED's
  ZIGBEE Board
    EFR32 MG1P132GI Blue Gecko Series 2 Bluetooth® Low Energy (SoC)
    I4BEB2 P1J552 EEPROM 
    TXC Oscillator T384 SMD 

PCB 2018-06-05
D-SS-X2-01-A-V2.1

A little different to this module https://github.com/basilfx/TRADFRI-Hacking

CableCatDK, 2019-10-29:
The Draco module is used in all new TRÅDFRI products. I have seen it in the new remote and in the signal repeater. When programming it there is no difference.

See also: https://wiki.iptables.dk/TR%C3%85DFRI_wall_switch#Extract_the_TR.C3.85DFRI_ZigBee_module

OneOfTheInfiniteMonkeys, 2019-10-29:
Is it possible to change the code on the E1745 TRÅDFRI motion sensor to have it work with Philips Hue?

basilfx, 2020-01-11:
It seems that this is the ICC-1-A module that I now have described in the README.md as well.

The only difference I found (so far), is that PF3 is now unavailable, and used to power the SPI NOR Flash.

closed 2020-01-11 by basilfx

EpicLPer, 2020-01-15:
What component actually controls the time setting on the E1745? I'd love to flash a custom firmware on that thing with just a 2-5 second "ON" state when there's movement, not 3 minutes minimum. I've already added the sensor to my Zigbee2MQTT network and read data from it via Home Assistant but the 3 minute timer doesn't allow me to play around with it much.
I've taken one of the sensors apart but I doubt I have the required hardware to read the contents of the Zigbee module... Not a "hacker" myself in that regard, max. I got is a serial to USB adapter :P