basilfx/TRADFRI-Hacking #28: Custom Full Stack

sorenfalch, 2019-11-30:
Hi Community!
Is there any work going on developing a full custom stack (eg. based on http://zboss.dsr-wireless.com/), a decompiled firmware stack, or otherwise? Would be fantastic to e.g. customize the RCU FW to control two groups of lights, etc (like two on/off remotes in one), etc.. make it control custom projects, etc...
Thanks 😊

basilfx, 2020-01-11:
Not that I am aware of, but that looks promising. I'll keep this issue open if anyone is working on this.

benemorius, 2020-02-11:
There's been at least some indirect work on this and it's actually not as far away as it seems.

I've been updating the zboss stack to support ZLL and Riot [here](https://github.com/benemorius/zboss). I've tested it with kw41z and at86rf233 and it's functional enough to do a zll pairing and control on/off/color/brightness. I need to get another sacrificial tradfri device to start testing it with efr32 which is what tradfri devices use but [RIOT-OS/RIOT#9212](https://github.com/RIOT-OS/RIOT/pull/9212) looks promising.

Basically a full stack replacement is a reality as soon as someone gets Riot's efr32 radio driver working with the zboss stack. It may be working already but I won't know until I test it.

Here's a demo so I don't look like I'm talking out my ass:
https://imgur.com/cdZd9je

kaibeckmann, 2020-02-11:
Thats sounds interesting. 
Do you use the RIOT at86rf233 driver / MAC layer or the one of the zboss stack?

If you are using RIOTs 802.15.4 layer, theoretically ZigBee and 6LowPAN could run in parallel (might be useful for dotdot).

benemorius, 2020-02-13:
It was easiest to just use the zboss mac for now. It sends packets by calling `gnrc_netapi_send()` with a full 802.15.4 header already filled, and it receives them by putting the interface into raw mode (`ifconfig 7 set raw`) and calling `gnrc_netreg_register(GNRC_NETTYPE_UNDEF)`.

Indeed the Zigbee stack and 6LoWPAN stack run just fine along side each other already. To accomplish that, I just added a hack to `gnrc_netif_ieee802154.c::_recv()` to bypass raw mode for the panid of my 6lowpan network and I suppose the better way of doing that would be to register each panid for whichever stack the packets should go to.

benemorius, 2020-05-28:
> Basically a full stack replacement is a reality as soon as someone gets Riot's efr32 radio driver working with the zboss stack. It may be working already but I won't know until I test it.

Turns out it's working already, though I have some zboss and riot bugfixes on the way to make it work better. Updates soon.

> If you are using RIOTs 802.15.4 layer, theoretically ZigBee and 6LowPAN could run in parallel (might be useful for dotdot).

I had to drop dual-stack on tradfri for now to stay within the 32k ram. It was only off by a few k and I already know a few places in zboss where I'm wasting lots of ram so I'm actually pretty sure that Riot can comfortably fit both zigbee and 6lowpan stacks on tradfri. I have no idea yet about BLE though.