basilfx/TRADFRI-Hacking #29: Pinout of module when using it as a remote

CableCatDK, 2020-01-12:
Here is the pinout of the 5 button remote.
[image]

Here is the pinout of the ON/OFF remote, if you flash its firmware to the module. The LED pin is not available.
[image]

Buttons are pushed by pulling the pin the GND with a 487 ohm resistor. Except paring, which is pushed by short to GND.

LED is powered by a 2N7002ET MOSFET.
[image]

basilfx, 2020-01-13:
Thanks for the information!

Can you confirm its part number? The [E1524](https://github.com/basilfx/TRADFRI-Hacking/blob/master/PRODUCTS.md#e1524-remote-controller) remote is already in the overview, but it seems that ikea now sells it as the E1810 (not sure though).

CableCatDK, 2020-01-14:
The new remotes contains the Draco module.

MattWestb, 2021-06-22:
Is some one knowing wot the TP1 and TP2 is connected to the module ?
I was hoping its the debug pins for SWD so not need using brute force for dumping the firmware but i have not testing it then i have not opening my remotes and dont like destroying them.