basilfx/TRADFRI-Hacking #32: Status of wireless peripheral support?

gnbl, 2020-01-21:
Is there any information about the support of the wireless functions (Zigbee for Tradfri, but BLE and others seem to be supported by EFR32MG1 as well) by your efforts (https://github.com/RIOT-OS/RIOT/blob/master/boards/ikea-tradfri/doc.txt) or the RIOT project in general?

basilfx, 2020-01-23:
There is some work done on the radio driver for the EFR32 for RIOT-OS: https://github.com/RIOT-OS/RIOT/pull/9212. It's based on RAILlib, that I presume is also used for the original firmware.

gnbl, 2020-01-23:
Thanks. That effort appears to have "died", so effectively there is no Tradfri radio support for RIOT and the hardware can only be used with the ARM MCU core.

kaibeckmann, 2020-01-23:
> Thanks. That effort appears to have "died", so effectively there is no Tradfri radio support for RIOT and the hardware can only be used with the ARM MCU core.

It's not completely dead. If you want to try it, I can rebase it to the current RIOT version. But be aware, there is one very annoying and persistent bug with the 6LoWPAN fragmentation... 

We also have a "version" for the NimBLE support of RIOT in a private repo, but currently no resources to bring it upstream to RIOT.

gnbl, 2020-01-23:
Thanks for the offer, but no need - unless there's an easy-to-use toolchain or a project to reuse the tradfri remotes as BLE controllers (Simplicity Studio code examples or generator), this is currently too involved for me.
I think this project's readme should point out that the hacking is somewhat limited.

closed 2021-01-11 by basilfx