basilfx/TRADFRI-Hacking #34: ICPSLC24-10NA in 30W dimmer

nica-f, 2020-04-24:
I just opened a 30W LED dimmer (the larger gray box) and found inside a ICPSLC24-10NA Rev1.0.0 module, which according to your findings would be the 10W LED dimmer. So either the -10NA can also do up to 30W or I got a very "very special" model from my Ikea shop :-)

CableCatDK, 2020-04-24:
It is the same module that is both in the 10W and 30W version. So if you just want the module, you can save money and buy the 10W version.

Se also this video: https://youtu.be/qQTJMTNVd40

nica-f, 2020-04-24:
Ah, good to know, thanks!
I just reported this as an issue since the repository with the pictures and schematics points out the -10 and -30 version as 10W and 30W...

closed 2021-01-11 by basilfx