basilfx/TRADFRI-Hacking #36: New GU10 bulb disassembling

MattWestb, 2020-06-29:
Have disabled one LED1875R5 without heating the lens with very small and hard flat screwdriver and working well and no cracking in the glass.
I have one LED1737R5 that was falling down and the glass its very broken so i want to change the glass with the LED1875R5. The problem was not getting the power pins losing. In the end I was using one gas heat gun and heating the connectors of the GU10 and trying getting the pins inside getting loose but was needing very mutch heating and the soldering on the PCB was loosing from the pins. Then taking the pins out from the GU10 connectors from inside. Resoldering the pins it's no problem but getting the PCB back in the glass with the pins in the GU10 connectors.
I can see on your photos than you have managed getting the PCB out from the GU10 glass without destroying the glass and PCB. 
It's the PCB pins soldered inside the GU10 connector or crimped in place ?
Then I have getting the LED1737R5 out without destroying it making it some more photos and dumping the firmware. and userdata and measuring the PSU voltage to the LEDs
Thanks in advance ! !
Mattias

basilfx, 2020-07-14:
The PCB pins are unfortunately crimped to the case. So it's quite hard to get them out. But it remains a cheap source to get the modules out ;-)

MattWestb, 2020-07-14:
Thank for sharing info !
Than I trying making the criping lager with a small steel neadel so i can putting the PCB from the LED1737R5 (with re solided pins) inside the good case.
Bay the way i having the one module up and running (in alpha state) as Zigbee coordinator under Tasmota after brickiing 2 ICC-A-1 modules with J-Link (bootloader writing app with wrong offset and corrupted peripheral and lockbits area).
I hope you and the other in the community its keep working on the EFR32 for the RIOT-OS project, its looks very interesting !!!

closed 2021-01-11 by basilfx