basilfx/TRADFRI-Hacking #39: New module in the GU10 RGB

CableCatDK, 2020-12-15:
IKEA product number: 804.392.28 (unrelased)
[image]
[image]

CableCatDK, 2020-12-15:
Text on RF shield:
MGM210L022INF2
Model: MGM210L22F
1934A01HB2

https://fccid.io/QOQMGM210L/User-Manual/User-Manual-4211003

[image]

[image]

CableCatDK, 2020-12-15:
The module can be bought here: https://dk.farnell.com/silicon-labs/mgm210la22jnf2/lighting-module-2mbps-3-8v/dp/3249880
Data Sheet: http://www.farnell.com/datasheets/2865974.pdf
[image]
[image]

basilfx, 2021-01-11:
Thanks for the heads-up. I have bought that light and torn it down. I'll add it to the repo.

basilfx, 2021-01-11:
I have added some information about this module and the product It is in. Thanks!

closed 2021-01-11 by basilfx

boelle, 2021-08-18:
@basilfx will check this, but did you add it as a kicad thing so we can use it without having to draw it up from scratch?