basilfx/TRADFRI-Hacking #47: possibility to emulate tradfri blinds

comlit, 2022-01-07:
hi
I recently connected my somfy rts blinds to my smart home. I can now control it through http requests. I got some Ikea smart home products already installed in my home, so I wondered if it is possible to emulate a tradfri blind and connect it to my existing work so that in the end it would be possible to control the somfy blind through the Ikea app. From what I saw in this repo it seemed as this would be possible. I would like to hear your thoughts and suggestions on this, since I have got no clue what I'm doing :). thanks

basilfx, 2022-01-12:
Emulation would mean that you would create some piece of hardware that behaves exactly as the IKEA blinds.

It would be much easier to see how the IKEA blinds work, and how you could hook it up to your existing blinds. So whenever the motor moves, route that signal to your existing blinds. You need to reverse-engineer the existing IKEA solution, to see what signals are sent to the hardware that physically controls the blinds.

Easy answer, hard(er) in reality ;-)

CableCatDK, 2023-05-23:
90% rabatt, that is something. I only got 32% rabat, and additional 200 DKK from a coupon.
[image]

MattWestb, 2023-05-23:
It was 50% off but only the battery was in the package no remote, mounting, changer instructions and so on and it was badly scratched but working OK.
I was asking for guarantee and the rest of the missing things and then was not saying and good things so i was saying its was for the trash but im interested in the Zigbee module and was getting one new price without guarantee.

honglihu1986, 2023-05-24:
[image]
[image]
[image]
[image]

MattWestb, 2023-05-24:
@honglihu1986 You must have some connection problem the the MG1P chip then it cant being debug locked only writ protecting flash pages.
Did you disconnect the motor and the battery so they cant interference with the debug connection ?
 Here is the `TREDANSEN block-out cellul blind` user data [E2022-UD.zip](https://github.com/basilfx/TRADFRI-Hacking/files/11550556/E2022-UD.zip) and it shall working with one normal IKEA bootloader plus rolling on extracted OTA file like we was doing with Symfonosk 2.

Also UD is only name is overrided and no hardware parameter so they is using the same motor controller board and software on it = very good.

honglihu1986, 2023-05-24:
Thank you very much for the guidance, I don't understand how to scroll on the extracted OTA file, I flashed this user data into an outlet, Home smart 1 shows a roller shutter, but it can't OTA into a blind, it shows the latest firmware.

honglihu1986, 2023-05-24:
[image]

MattWestb, 2023-05-24:
Ops its little more advanced.
The OTA file is rapped with metadat that need being removes and you is getting one EBL/GBL file that can being flashed with commander.
I must look for the ext instruction but you can doing it with zigpy.cli.

If you have dumping the main flash of one blind its the easiest way only flashing it on one new module and only need the user data for working OK and can being OTA updated without problems.

MattWestb, 2023-05-24:
For dumping MG1P modules you can using my old guide that is not complete but is working and explaining most things https://github.com/MattWestb/IKEA-TRADFRI-ICC-A-1-Module/tree/master/Flashing-MG

MattWestb, 2023-05-24:
By the way one updated firmware for all blinds in in the pipe.
https://csa-iot.org/csa_product/connected-roller-blind/