basilfx/TRADFRI-Hacking #50: Converting ICPSLC24-10NA to 12v

Treantno, 2022-03-18:
I have been hacking around converting to make it run on 12v. With a 12v input and light load it sort of works. But a log led strip would not turn on. Changing R9 from 270k seems to do the 12v conversion trick, for now I have added a 230k in parallel (making it 124.2k). R9 should ideally be 125k for a 12v conversion.

pkonowrocki, 2022-10-05:
@Treantno What does "sort of works" mean? And how long does the strip have to be to not turn on?

Treantno, 2022-10-05:
I have not experimented with different lengths. The issue is that the overload  protection is based on detecting low input voltage to the switch, and if it's set just bellow 12v and with light load it will turn off due to low input voltage. The resistor change lowers the threshold to work fine with 12v input.