basilfx/TRADFRI-Hacking #53: Is network cloning possible from gateway?

vuokko, 2023-06-25:
Hello,

My current situation is that my old tradfri gateway is dying. It has had twice open surgery on my bench.
The other problem is that there is no migration path. Re-pairing everything would be irritating task.

I'm probably going for SMlight's slzb-06 and home assistant, so this could be little bit offtopic.

Everyone is repeating the gospel, migrating without re-pair isn't possible, but no-one isn't really telling the reasons. Is here someone who can give some answers why it wouldn't be possible.

My limited understanding is that with some steps it is possible. 

1. Join to the network
2. Learn long and short network names
3. Map the network devices
4. Ask the direct communication keys to all found devices from key store. This may be something where I missed something important.
5. Copy network settings and keys to our new gateway
6. Pull the plug of the old one
7. Turn on the new one. 

Is there something I missed? Another protocol on top of the zigbee which makes life hard?

MattWestb, 2023-06-26:
The TF network working mode is one Zigbee Light Link = every controller is responsible for adding devices and allocation of network addresses and groups and the TF GW is one _Zigbee Controller bridge_ as device type = ti can communicating with all devices in the network and also controlling them but cant manager the network security then its coming from the network was started and cant being changes and if it was you need changing it in every device routers and controllers that is not possible.

DIrigera, Z2M and ZHA is running in Zigbee 3 centralized security mod (Trust center) with one coordinator that is manager the security and all is centralized so you is having most of the information of the network in the coordinator and can also manage the network from it.

You can adding one device with classic paring to TF GW and getting the network credentials (its how to getting the network key for sniffing with wireschark) but you cant changing the network parameters in all devices have getting then joined the ZLL network then it was sett with touch link commands then it was joining it and can only being sett then joining the network.

Not working parameters that you cant changing in device is bindings for reporting and also network addresses and group you is gonna getting conflict in the future that all devices is have there range set with TL and cant being changed. 

You need reading some papers how ZLL / Zigbee 3 network forming and security is working then its complete different methods and cant being changed on the fly.

Silabs is having some good ones but i dont remember the name of them.

Recommendation set up / forming one new network with your TI coordinator and adding routers from near it and out and then adding end devices to the ideal parent routers and you is getting one good mesh network.

vuokko, 2023-06-26:
Thanks for a good answer. I didn't know about ZLL before and learned a lot already. And now I know what to study and maybe learn a way how to work with both of them.